Ai-Thinker-Open/GPRS_C_SDK #446: Connecting module to car battery

marcosasilvalepe, 2020-08-05:
I've had this module for a couple of months now and it works like charm! 

I have it making  POST requests to my website using the gps_tracker demo with a few alterations that I made to the code but the problem I have is that whenever I start the car it takes several minutes before it can actually get a gps fix. 

The GSM part works right away but the GPS takes a while so I'm thinking maybe I should connect the module to the battery and  put it to sleep when the car gets turned off, so that way maybe it takes less time to get a fix in the morning.

I don't have a lot of experience working with microcontrollers. I'm basically a self-taught web developer so Javascript, PHP, Node JS, etc, I'm good at but I don't exactly know how to make the right connections in order for this to work and not burn the module. 

I bought a 12v to 5v-3amp converter and  based on this #129  I should connect the +wire from the car battery to both the VBAT (Pin 17) and to the VUSB/5V Pin ?

Any help would be greatly appreciated!

punee995, 2020-08-06:
The A9G module supports only 3v5 to 4v2(idle value is 4v0) on VBAT pin, so if your module has little pot on board you can change it to 4.0v and then you're good to go.

marcosasilvalepe, 2020-08-06:
OK, thanks. 

So I've read a bit on how to step down from 5v to 4v and I think that what I would have to do is connect the 12v to 5v converter to the car battery and then, with either silicon diodes or resistors, step down the 5v to 4v and plug it in the VBAT pin, and I could read the voltage to trigger an event to make the module go to sleep if it's 4volts and wake up if it's 5volts. 

The module would be connected to the car radio I think, that way whenever the car starts the module should get the 5volts.

But what happens when the car starts ? The module would burn by receiving 4 volts from the VBAT pin and 5 volts from VUSB/5V at the same time ??

zkemble, 2020-08-06:
I'm guessing you're using the pudding dev board, right? VBAT is already powered by 4.2V from an onboard 5V to 4.2V regulator. VUSB is not a power supply pin, it just monitors for USB power and turns on the module on.

What you need to do is find a power connection in your car that is always powered, even when the car is off. I think the OBD-II port is always powered with 12V. Now that your module is always powered you then need to add some code to monitor a 12V supply that only turns on when the car is on. The other 12V supply can't be directly connected, you need a [voltage divider](voltage divider). Vin to 12V, Vout to an input pin on the module. 10k and 3.9k will be suitable values for the resistors. When the pin reads HIGH, the car is on. When LOW, the car is off.

Now, the actual power saving. The GPS receiver must always be enabled, otherwise the position will be lost. There's not much documentation for the GPS receiver in the A9G, but there does seem to be GPS_SetStandbyMode() and GPS_SetLpMode() functions in https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/libs/gps/src/gps.c and https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/libs/gps/include/gps.h

You should put the GPS receiver into low power modes and disconnect from the GSM network when the 12V monitoring pin reads LOW. When the pin is HIGH you then reconnect to GSM and put GPS into full running mode again.

But be warned, the module is very unstable, it might only last a few days before it crashes.

marcosasilvalepe, 2020-08-07:
Cool! I'll have to read up on your response because like I said I'm a noob in microcontrollers but now I know which way to go. It will take me a couple of weeks to get this going.

I haven't had any problems with the GSM network. The module connects very fast and it starts transimitting the data without problems. It's the GPS getting a fix that's my current problem. It didn't had that many problems a month ago or when I stopped using it that much to start working with the Google Maps API, but now after not using it for that period it struggles to get a GPS signal. Maybe it's an antenna problem so I'll try to buy a bigger one and see if that solves it.

Anyway thanks for response, I'll post the solution once I implement it based on the info you gave me.

vkhungund, 2020-08-07:
hi @marcosasilvalepe , 
why do you want to connect it to car battery?Theres always a risk of power loss may be during cranking,
Instead try to hook up a ~1000mAh po bLiPo battery(make sure they stay cool in your car as they might burst!).
I recently bought this Module and am also trying to buid a car GPS tracker with MQTT.

Interested in your progress.

marcosasilvalepe, 2020-08-07:
Hi @vkhungund I just don't want to lose the GPS signal because it takes too long for it to get a fix again once the module restarts.

Like I said I'm using the gps_tracker demo to make post requests with latitude, longitude, speed, timestamp and other variables to a php script in a website that captures them and inserts all the values in a MYSQL database, and on the client side I have a Javascript function that runs every 2 seconds and fetches the last row of the table with the latest values to slide the marker to the new position on the map. The module handles itself very well and it was pretty stable about a month ago but like I said I stopped using it to work on the Google Maps API and now for some reason it struggles to get a GPS fix.

I had actually bought a couple of 3.7v 500mah Li-ion batteries but because of covid-19 it has taken longer than usual for them to get here so I have to wait to keep playing with the module but I'll post when I make some progress. Cheers.

marcosasilvalepe, 2020-11-05:
So to do a follow up, based on the answer by @zkemble (thanks for that) I ended up with this

[image]

It looks pretty amateur but like I said I'm a noob in microcontrollers and it works.

I checked the cables from the car radio and it turns out that there is a cable that is always powered with 12V so that the radio doesn't get disconfigured everytime you turn the car off, and another 12V cable that turns the radio on when the car is on.

So in the image the orange cable only gets power once the car is on, and the brown cable is connected to a 12V to 5V converter, that way the module always has power.

I switched to pulkin micropython version to program the module and by simply doing `pin.value()` you get either 0 when the car is off and that way the module needs to go to sleep, or 1 when it's on and it should transmit coordinates through the internet.

I'd like to install a small battery to the module now just to provide power in case it can't get it from the car battery, like if for instance someone stole the car and cuts the module's power cable, that way the module can transmit coordinates anyway at least for a few minutes or hours, but I don't know how to do that with my current configuration. 

There are 2 VBAT pins in the module and if I connect a battery while getting power from the 5V cable at the same time, wouldn't it burn the module because it get's power from 2 different sources ??