Ai-Thinker-Open/GPRS_C_SDK #447: Burning firmware to a9g module

pagman, 2020-08-13:
Hello! I have finished testing my code with the pudding board and now I am designing my custom PCB for my project but I have a problem where I can it find the answer.
I will power the module with a lipo 3.7v  from the vbat pins and the appropriate decoupling capacitors and I will connect the hst_rx/tx pins to my ftdi programmer with 3.3v logic level pins the same as I used to do with my pudding board. Do I need any resistors in series with my tx/Rx pins to burn the firmware?

Thank you

zkemble, 2020-08-16:
The pudding board doesn't have any resistors on the HST pins, so no need to have them on your custom board! But, if you're worried about the voltage differences then maybe a 680R resistor on the HST RX -> TX serial converter line will limit any backfeeding currents to safe levels.

pagman, 2020-08-16:
Thank you very much I appreciate your help

closed 2020-08-28 by pagman