Ai-Thinker-Open/GPRS_C_SDK #448: GPIO in A9G

shmusb1999, 2020-08-16:
Hi 
I'm working on A9G.
I have two questions:
1. after activating all the IO pins, I see that the voltage of all IO pins are 2.8V while we expect IO8~IO24 have to be 1.9V!
why does this happen?

2. I don't understand the functions of OS_SetUserMainHandle and OS_CreateTask completely. why OS_SetUserMainHandle should not be used after OS_CreateTask sometimes?

zkemble, 2020-08-16:
1. The docs do say that IO8 - 24 are 1.9V, but it's wrong, all pins are 2.8V.
2. OS_SetUserMainHandle() sets the task that receives system events, like API_EVENT_ID_POWER_ON and API_EVENT_ID_SIGNAL_QUALITY. https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/os.html#ossetusermainhandle

shmusb1999, 2020-08-24:
> 1. The docs do say that IO8 - 24 are 1.9V, but it's wrong, all pins are 2.8V.
> 2. OS_SetUserMainHandle() sets the task that receives system events, like API_EVENT_ID_POWER_ON and [...]

Thanks so much

I had studied the link you send before but I steel cannot understand.
In some codes, there is OS_SetUserMainHandle() once and OS_CreateTask() more than once. why does this happen?

zkemble, 2020-08-24:
There can be multiple tasks running at the same time (from OS_CreateTask(), kind of like a thread), but only one of those tasks can be the main task.