Ai-Thinker-Open/GPRS_C_SDK #454: CSDTK links not working (direct link, mega, baidu cloud)

ece-mohammad, 2020-09-03:
## 1. SDK version（SDK 版本）

{
CSDTK 4.2 ($Revision: 43828 $)
GPRS_C_SDK_V2108
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
As the title says, the CSDTK4.2 links in github releases, and SDK documentation wiki are not working. The file is no-longer available.
}

---

Iteinos, 2020-09-05:
Same here pls resolve asap

wzab, 2020-09-05:
I confirm. Tools are not available. I can't start working with the board :(

nguyenvanduocit, 2020-09-09:
Anyone have downloaded the file?

zkemble, 2020-09-09:
Check here! -> #412

nguyenvanduocit, 2020-09-10:
@zkemble thanks

ece-mohammad, 2020-09-11:
@zkemble Thanks

closed 2020-09-11 by ece-mohammad

nguyenvanduocit, 2020-09-14:
@ece-mohammad Why did you close the issue without fixing? Link to the fork does not fix. Or we need to move the issue to repo GPRS_C_SDK_DOC?