Ai-Thinker-Open/GPRS_C_SDK #455: A9 serial first-time output wrong characters

glararan, 2020-09-06:
Hi,

Few months ago I had issue #428. Finally I had time to solder it and get it working.
My schema is following (bold X => not soldered yet)

[image]

When I power up board green diode (red box) blink twice. Why? Its VSIM :O

I measured voltage. 5V node is 5.1V, 4V node is 4.15V. A9 is powered on 5V pin by 5V, from VBAT 4.15V. 

I am using serial-to-usb CP2102 module 
[image]
https://www.ebay.com/itm/CP2102-MICRO-USB-to-UART-TTL-Module-6Pin-Serial-Converter-STC-Replace-FT232/193275338005?epid=1064094607&hash=item2d001b9d15:g:TYMAAOSwPGBeBuim

My AI-Thinker Serial Tool v1.2.3.0 configuration + output
[image]

Any idea what I should focus to get it working? Then I would follow with burning.

zkemble, 2020-09-07:
Well, you told use what it does, but are they actually problems? What were you expecting the green LED/VSIM to do? Does yy in serial conconsole only happen at power up (which is normal), does it stop you from programming it?

glararan, 2020-09-07:
Oh sorry if I wrote it down bad.

Problem is that if I send over serial port command `AT `then I receive `yy` instead `OK`.

I would expect from green LED on VSIM pin turned all time? If I measure VSIM its 0. Shouldn't it be around 3V? 

I didnt try programming as its not recommended if I dont receive `OK `in console by article http://acoptex.com/project/15083/basics-project-101a-ai-thinker-a9g-gprs-gps-development-board-at-acoptexcom/

zkemble, 2020-09-07:
Are you sending AT to UART1/2 RX/TX? The HST port is only for programming. VSIM is probably only powered when accessing the SIM card so reduce power consumption. I think the A9 supports 1.8V and 3V SIMs, so the VSIM voltage will also depend on the SIM card.

glararan, 2020-09-07:
@zkemble yeah.. you were right... I didnt know about HST not supporting it. It works. Thanks

closed 2020-09-07 by glararan