Ai-Thinker-Open/GPRS_C_SDK #456: Unable to download program via coolwatcher

natrayanp, 2020-09-07:
> fastpfGo()
Fastpf V2 over Host
Loading the lod files: 
sms_B2129_debug.lod
Using flash programmer: 
host_8955_flsh_spi32m_ramrun.lod
Ramrunning the flash programmer...
Event sniffer (re)starting. (Connection: Event Sniffer (COM1))

 undefined method `CHST_Read32' for #<CHHostConnection:0x7f768d4792c0> (NoMethodError)

./rbbase/common/base.rb | 85 | `read'
-- | -- | --
./rbbase/common/base.rb | 239 | `read'
./chipgen/Modem2G/toolpool/scripts/8809/target_executor.rb | 72 | `enterHostMonitor'
./chipgen/Modem2G/toolpool/scripts/ChipStd/target_executor.rb | 366 | `session'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 12 | `doramrunXfer'
./chipgen/Modem2G/toolpool/scripts/ChipStd/ramrun.rb | 62 | `ramrunXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 212 | `dofastpfXfer'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 718 | `dofastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpf.rb | 788 | `fastpf'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 83 | `fastpfGenericGo'
./chipgen/Modem2G/toolpool/plugins/fastpf/fastpfplugin.rb | 112 | `fastpfGo'

EVENT SNIFFING ERROR: CONNECTION BROKEN?
EVENT SNIFFING RESURRECTED.

I am using Linux Ubuntu.  How to resolve this issue?

ens4dz, 2020-09-24:
Did you flow the Documentation ? :
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/burn-debug.html

> Module Debugging The download interface is the HST interface. The pinsHST_RX and HST_TX are respectively connected to theTX and RX pins of the USB to serial module (cross connection,HST_RX then TX)
> 
> ...
> Create a link file for serial port as follow, and later you can fill in the coolwatcher's chart with 1,the COM1 must upper-case
> sudo ln -s /dev/ttyUSB0 comport/COM1
>

TNeutron, 2022-06-30:
Connect to HST_RX/TX, I found that works and solves this issue
.

machiobrian, 2024-01-28:
How have you been working, if you just found that out?😂😂😂😂