Ai-Thinker-Open/GPRS_C_SDK #457: SD Card sometimes not mounting

ihewitt, 2020-09-16:
## 1. SDK version（SDK 版本）
{
csdtk42
SW_V2129_csdk
}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
{

Frequently after resetting the module the SD card doesn't appear to have been mounted successfully, any file API_FS operation into "/t/" fails, the only way to resolve this is to remove power completely from the board and restart it.
Simply resetting, either using a software PM_Restart, or using the hardware switch is insufficient to resolve this.

Is there any way to either force a full restart similar to cycling power, or, any way to re-mount the SDCard in software?

(Initially I just assumed this was an SDCard make issue, so have alread tried with multiple manufacturers and sizes (from 2Gb to 16Gb) with no luck)

}

---

zkemble, 2020-09-16:
Could be related to this #421 - The A9G eats a lot of power on the 850/900MHz GSM bands which can cause voltage drops. These drops can mess up the SD card and require a power cycle.

ihewitt, 2020-09-16:
Hi thanks for the speedy comment! yeah I saw the comment about GPRS power in the various demo apps, and I've tried lots of combinations of settings, turning things off, frequencies etc.... but this seems completely unrelated to anything. :(
i.e. when I run connected to the debug pins, even before my app does any connection or GPS calls, if I just have it flashing lights and "walking the SD card on startup to read files" sometimes on powerup all the File IO calls fail until full power off.
Noticed while developing as I re-flashed test code over and over, after two or three restarts it would die.
Annoying - it would be useful to be able to recover the situation without needing physical power off. Guess I'll have to re-think/rework my little project to run without relying on SD at all.
Or consider a GPIO controlled power switch! :grin:

ihewitt, 2020-10-04:
You were indeed 100% spot on! after adding your capacitors to my design,
[image] it's never had an SD card failure since.
I have different issues with network registration now! :roll_eyes: it seems roaming between networks (I think) causes trouble. I'll need to crank up the debugging to investigate.

ekostero, 2024-02-15:
Hi all, I have a a9g module and I use micropython in it. Can somebody to give me an example how to mound or use it
Thanks