Ai-Thinker-Open/GPRS_C_SDK #460: Save Latest Firmware of A9G

vkhungund, 2020-10-04:
Hi,

I have an A9G module which has below firmware version.
```
AT+GMR

V02.02.20190915R
```
Before starting to flash my firmware, i was curious if i could somehow save the complete firmware in case i want to revert back to original factory firmware so that atleast i could use the module using AT commands on UART?

I figured out we could flash Factory/Dev firmware using "fpugrade" tool but the lastest [available firmware](http://wiki.ai-thinker.com/gprs/firmware) is of V02.02.20180825/20181011 version and there is not a single info on the current firmware(20190915) that is flashed on the A9G.

I dont know why Ai-thinker is not updating the info properly.
The HW specs are really interesting but finding the toolchains seems to be really daunting due to missing links:-(.

Thanks!

ihewitt, 2020-10-04:
OK, it's possible. but not straightforward. basically you need a "save flash to SD Card" utility, then, you need to "ramrun" it from coolwatcher without flashing it, I have a some code to do it (I needed to restore an A9G accidentally-completely-wiped during experimenting with random calls). It's late here at the moment, but I'll dig it out and post it here.
I'm not sure if there's an easier option or utility that does this already, I'm sure others will mention if so.

vkhungund, 2020-10-06:
Hi @ihewitt , 
 
do you mean inserting a SD card into the A9G module ? 
Could you share the documentation/code about "ramrun"?

Thanks for your hint!

closed 2020-10-06 by vkhungund

ihewitt, 2020-10-06:
Ah I was getting muddled with one of my other devices... hmm, I'm not sure if there is an equivalent of "run without burning" in coolwatcher.

anyway.... this code: https://gist.github.com/ihewitt/7ef825261cc642398cf795f394af7539
When burnt will dump the contents of the flash onto the SD card, although when burnt you will lose any "APP" already installed, i.e. you wouldn't get a copy of the default AT app, but this will take a copy of the platform and bootloader (i.e. base SDK). Perhaps anyone else can comment/suggest if it's possible to run without burning?

ihewitt, 2020-10-06:
haha spoke too soon, ok I think I've got the binary protocol to query the memory. I'll see if I can write a command line "dump" via HST app.

vkhungund, 2020-10-06:
How did you find the HW protocol? 
I'm trying to figure out how coolwatcher is communicating on the COM port using serial sniffer.

let me know if you have any update, I'm curious.

ihewitt, 2020-10-07:
On Tuesday, 6 October 2020 21:40:00 BST ViHu wrote:
> How did you find the HW protocol?
> I'm trying to figure out how coolwatcher is communicating on the COM port
> using serial sniffer.
> 
> let me know if you have any update, I'm curious.

In coolwatcher you can open up the "coolhost" window... in there you log and 
view the serial I/O there's a "save com data" on the "other" tab which you can 
*then* use to open up and view the traffic.

I've got a small basic serial app that can read the flash now.... although I'm 
not 100% sure I've got the serial configured quite right since it dies after 
some I/O has gone through. Will play with it a bit more after work and paste 
to gist.

ihewitt, 2020-10-07:
Here's a gist of what I'm trying to get working.. proof of concept:
https://gist.github.com/ihewitt/5969b7d427fc7248306cb894ec20cace 
almost works. needs a fresh boot of the A9 will start to dump flash contents (of the app section - https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/zh/more/flash_map.html)

 but I'm not a serial expert and I can't figure out the right way to get the protocol working. but it starts to work (on a fresh booted A9G) but gets checksum errors and breaks.... :(

ihewitt, 2020-10-12:
Done. Proof of concept. Needs editing to extract different parts of flash (and its slow) but can be used to extract all firmware sections without flashing anything.
https://github.com/ihewitt/ivrtrack/blob/main/util/dump.c

vkhungund, 2020-10-15:
Found something interesting ;-)
[image]