Ai-Thinker-Open/GPRS_C_SDK #461: SD/TF Card Filesystem

Alizame, 2020-10-09:
What Filesystem should be used?
I'm using a 2GB SD-Card and I tried a few different ones, but i couldn't get it to work.

I always get an error from the fs demo:
> ...
> Get FS T Flash device info fail!
> T Flash used:0 Bytes, total size:0 Bytes(0.000 MB)
>...
> Start Fs  TF card read write test!
> Open file failed:-1

ihewitt, 2020-10-10:
Standard FAT32 *should* be fine. There are power issues driving the SDCard as the board powers up GRPS though: https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421 so if you're getting  intermittent errors try 1. making sure to have enough good power and/or 2. power cycling the board (not just resetting it).

Alizame, 2020-10-11:
Hi, thanks so much for the help so far!

I looked at #421 (and #457 ), but I don't have the problem that it "sometimes" doesn't mount, it never works/worked ;(
I always power cycle it between tests.

I supply the pudding board with my bench PSU (has a 5A current capability) at the VBAT-Pins.
(I also looked if I have some fitting caps, but no luck so far, i may need to order some LOW ESR ones tomorrow)

I tried FAT32 with allocation unit sizes of 4096 or 512 bytes, both not working (i formated the sd-card via windows).

On another note: does someone know why the altitude is 0.00000 in the GPS example?

mirekhk, 2021-04-30:
Hi all,
I have same problem. The SD card is not mounted at all even I tried as many as possible FAT versions. Can someone help here?

Thank you

ermito, 2023-06-04:
Same problem. tried different cards