Ai-Thinker-Open/GPRS_C_SDK #464: Error n°14 return from AUDIO_RecordStart()

alepdb, 2020-10-27:
## 1. SDK version（SDK 版本）

{

CSDTK42

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
I connected the ADC of the A9G device to a sound sensor: when it hear a sound, A9G starts an audio recording.
For the first 6 recording there is no problem, but at the seventh audio recording, the function AUDIO_RecordStart(AUDIO_TYPE_AMR,AUDIO_RECORD_MODE_AMR122,recordFd,NULL,NULL); return the error n°14 (AUDIO_ERROR_FILE_OPERATE), and then is impossible to create/open other files.
I attached my code.
[demo_recording.txt](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/5447404/demo_recording.txt)

}

---

Yakir-Ben-Hillel, 2021-03-03:
same bug here

alepdb, 2022-03-31:
@Yakir-Ben-Hillel did u solve it in some way?