Ai-Thinker-Open/GPRS_C_SDK #465: Could somebody review my schematic please

bokolob, 2020-10-29:
It's a GPS tracker with internal  gps antenna,  accelerometer,  attiny, battery charger.

[image]

(it's an open project if it matters to you :)

Thanks!

## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

zkemble, 2020-10-29:
What are all the things you are expecting it to do? The schematic is rather spread out and difficult to figure out what is connected to where (especially all the BAT_IN, V_BAT, 5V lines), but from a quick look it appears that when you plug USB power in the tracker will turn off, is that what you want it to do? What are D3 and D7 supposed to do?

bokolob, 2020-10-29:
When USB is connected, the device is powered by buck converter U8 (5 to 4 volts) and the battery is charging with U4.
When it is disconnected the Q3 mosfet switches power to battery. 

V_BAT - battery connector's line.
5V - USB power line,
BAT_IN (yep, naming is awful) is an internal power line which is commutated either to USB or battery.

I'm not sure that d3 is needed, but it prevents q3's gate from getting a charge from the C6 capacitor.
d7 is needed to block buck convertor back-feeding and battery discharge through R13-R14 divider. 

I will use software I2C and SPI, so pins don't matter

zkemble, 2020-10-29:
When USB is plugged in the buck regulator will be connected to the battery through the internal diode of Q3, allowing uncontrolled current to flow into the battery until it reaches around 4V. Have a look at some kind of load sharing thing like [this](https://blog.zakkemble.net/a-lithium-battery-charger-with-load-sharing/), you kind of almost already have it with Q3, mainly need to flip it around.

R2 and R4 are not needed, the A9G already can already read its supply voltage with PM_Voltage().

bokolob, 2020-10-29:
Hm, why do they have R2 and R4 on the pudding board?

bokolob, 2020-10-29:
Wow, is this parasitic diode so important? Thanks!

bokolob, 2020-10-29:
Could you explain your scheme please? It seems that GS voltage will be always 0-0.25V (diode drop)? Which is not enough to open  fet.

zkemble, 2020-10-29:
> Hm, why do they have R2 and R4 on the pudding board?

¯\_(ツ)_/¯

> Wow, is this parasitic diode so important? Thanks!

Yup, it's pretty much just an extra diode across the MOSFET and must be treated like one!

> Could you explain your scheme please? It seems that GS voltage will be always 0-0.25V (diode drop)? Which is not enough to open fet.

When USB 5V is removed gate will be 0V, source will be VBAT (through internal diode), so Vgs = (0 - 4) = -4 = Open.

bokolob, 2020-10-29:
Ok, I've fixed it :) Are there some other issues?