Ai-Thinker-Open/GPRS_C_SDK #467: GPIO input mode - default high level not working

glararan, 2020-11-13:
## 1. SDK version（SDK 版本）
V2.108

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）
Pin is initialized as
```
GPIO_config_t gpioSwitchOne = 
    {
        .mode         = GPIO_MODE_INPUT, 
        .pin          = MAGNETIC_SENSOR_PIN, // IO20
        .defaultLevel = GPIO_LEVEL_HIGH,
        .intConfig.debounce = 0,
        .intConfig.type = GPIO_INT_TYPE_MAX,
        .intConfig.callback = NULL
    };

GPIO_Init(gpioSwitchOne);
```

And suppose to have high level by default however I can measure about 20mV. 

Reading:
```
        GPIO_LEVEL switchOne = 0;

        GPIO_GetLevel(gpioSwitchTwo, &switchOne);

        Trace(1, "Door: %d", switchOne == GPIO_LEVEL_HIGH ? 1 : 0);
```

zkemble, 2020-11-13:
`.mode         = GPIO_MODE_INPUT,`
Looks like you've got it set as an input? HIGH/LOW level is for driving it as an output.

glararan, 2020-11-13:
@zkemble that's correct. I set it as input.

I thought setting defaultLevel might work as pull-up? Nope?

zkemble, 2020-11-13:
No unfortunately, there’s no support for enabling any internal pull-up and pull-down resistors :(

glararan, 2020-11-13:
oh ok, thanks for info.

closed 2020-11-13 by glararan