Ai-Thinker-Open/GPRS_C_SDK #470: AT commands stack and SDK source code

HrMitrev, 2020-11-20:
Hi!

We are building a custom device which is based on GPRS connectivity. A while ago we saw this repo and while thinking the SDK is open source and the AT stack being the same, we switched from SIM800 to A9. Recently however after trying to use A9 without the AT stack, we realized that the lack of source code and proper documentation of the APIs and the RDA8955 SoC, the device is pretty much unusable. If using the AT stack, the device is problematic at best. So we're loosing trust and probably will switch back to SIMCom or something else.
I'm telling you this not that I expect you(or RDA) to change something, (even the GNU tools' souce code is not publicly avaliable which is a violation of the license!) but to suggest why this device which has the potential to be as awesome, big and popular as esp32 is so much less adopted.
Going forward, I suggest to you to rethink the strategy you're using. Maybe try to talk to RDA.
If we had the source of the AT stack we would have been able to add our custom commands to better utiliaze the peripherals of the chip. If the SDK was open source and there was a technical reference manual for the SoC we would have had the trust to implement our own AT stack from scratch.
And to finalize my point - the last A9G pudding boards we bought were with firmware version : 
A9/A9G
V02.02.20190915R
Ai_Thinker_Co._Ltd.
Which is not even avaliable for download as binary file(the latest one is from 2018) from your website!!! Not to mention there were breaking changes...

bokolob, 2020-11-20:
You had better write an email to their support... I bet - nobody will answer here.

BTW, did you try embedded mode on SIM800? Is it possible to port micropython there?

HrMitrev, 2020-11-21:
@bokolob I have not tried to write firmware for SIM800 and I don't know if it's possible.

Also at this point I don't care if they answer or not - it's their own company business

LubomirBogdanov, 2020-11-25:
Yes, if Ai Thinker are able to share the code for the SDK that would be fantastic!

It is a great product and it would unleash a lot of features!

XaviDCR92, 2020-11-26:
I also asked Ai-Thinker to release the source code of all GPL-licensed tools via e-mail and also on issue #431 with no response so far, so we the community have decided to reverse-engineer both the RDA8955 and the SDK. Discussion is active on issue #458.
I am also writing notes and tools that help me in the process at https://gitea.privatedns.org/Xavi92/a9_free/

faithware, 2021-01-18:
Even using the SDK the device has known issues and I do think that you need to switch back to simcom or try MC60 which is programmable. 
I lost 2 years developing on this A9 and I am in real trouble with my customers. 
Cheers