Ai-Thinker-Open/GPRS_C_SDK #475: The download links of CSDTK 4.2 are not available!

nimamgs, 2021-01-24:
Hi Guys,
Please check the download links of CSDTK 4.2.
NONE of them are available!!!

 http://test.ai-thinker.com/csdk/CSDTK42.7z
https://pan.baidu.com/s/16ZIQ9YbZdcVz-zvpzEFXbA
https://mega.nz/#!uKAlRA7Z!UPHn3ZoLY8X33iMIKcBcUYFwrhmNYe-x_dyOdrm_d6I

Thanks in advance

## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

syed6996, 2021-10-11:
Hello guys,

These links are still not available to download CSDTK4.2 

https://pan.baidu.com/s/16ZIQ9YbZdcVz-zvpzEFXbA
https://mega.nz/file/uKAlRA7Z#UPHn3ZoLY8X33iMIKcBcUYFwrhmNYe-x_dyOdrm_d6I
http://test.ai-thinker.com/csdk/CSDTK42.7z

Is there any other ways to download the CSDTK?

@nimamgs did you find solution for the issue.

I am trying to install GPRS_C_SDK in A9G GSM/GPRS module. 

Referring this document :
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html

Thanks in advance.

gfrodo, 2021-12-30:
Hi, I downloaded CSDKT4.2 from here: https://github.com/pulkin/csdtk42-linux

Unisolgr14, 2022-02-01:
Hello friends 
Have you got the CSDTK ? Please share it.

fflorey, 2022-04-02:
Same problem here: Following the instructions on 

https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html

The CSDKT4.2 is not available... PLEASE HELP!

sake402, 2022-06-09:
Hi. I found a copy here. Hope it helps
https://gitlab.com/keery-portfolio/a9g-sdk-ception

niwantha33, 2022-06-21:
Hi, You can use this also,  https://github.com/niwantha33/CSDTK42.git 
I got it from Ai-Thinker developer in Sep 2020.

radar1040, 2023-01-17:
Thank you friend.