Ai-Thinker-Open/GPRS_C_SDK #478: program crash - memory leak?

glararan, 2021-02-05:
Hi,

I am using simple program to try circut. 

Source code:
```
#include <api_os.h>
#include <api_debug.h>
#include <api_call.h>
#include <api_audio.h>
#include <api_event.h>

#include <ctype.h>

#define MAIN_TASK_STACK_SIZE    (2048 * 2)
#define MAIN_TASK_PRIORITY      0 
#define MAIN_TASK_NAME         "Main Task"

// CONFIG
#define CONTACT_NUMBER		"my contact number" // e.g. 123456789

static HANDLE mainTaskHandle = NULL;

static uint8_t flag = 0;
static uint8_t phoneFlag = 0;

static uint8_t first = 0;

void EventDispatch(API_Event_t* pEvent)
{
    switch (pEvent->id)
    {
    case API_EVENT_ID_NO_SIMCARD:
        Trace(10, "!!NO SIM CARD%d!!!!", pEvent->param1);
        break;

    case API_EVENT_ID_SYSTEM_READY:
        {
            Trace(1, "system initialize complete");
            flag |= 1;
        }
        break;

    case API_EVENT_ID_NETWORK_REGISTERED_HOME:
    case API_EVENT_ID_NETWORK_REGISTERED_ROAMING:
        {
            Trace(2, "network register success");
            flag |= 2;
        }
        break;

    case API_EVENT_ID_CALL_DIAL:
        {
            phoneFlag = 1;

            if (pEvent->param1)
                Trace(1, "Dial success");
            else
                Trace(1, "Dial failed");
        }
        break;

    case API_EVENT_ID_CALL_HANGUP:
        {
            phoneFlag = 0;
        }
        break;

    case API_EVENT_ID_CALL_INCOMING:
        {
            Trace(1, "Incoming call...");

            if (!CALL_HangUp())
                Trace(1, "Failed to hangup incoming call");
            else
                Trace(1, "Hanged up incoming call");
        }
        break;

    default:
        break;
    }

    if (flag == 3 && first == 0)
    {
        first = 1;

        if (!CALL_Dial(CONTACT_NUMBER))
            Trace(1, "DIAL FAILED");
        else
            Trace(1, "DIALING");
    }
}

static void MainTask()
{
    API_Event_t* event = NULL;

    while (1)
    {
        if (OS_WaitEvent(mainTaskHandle, (void**)&event, OS_TIME_OUT_WAIT_FOREVER))
        {
            EventDispatch(event);

            OS_Free(event->pParam1);
            OS_Free(event->pParam2);
            OS_Free(event);
        }
    }
}

void nedoma_Main(void)
{
    mainTaskHandle = OS_CreateTask(MainTask, NULL, NULL, MAIN_TASK_STACK_SIZE, MAIN_TASK_PRIORITY, 0, 0, MAIN_TASK_NAME);

    OS_SetUserMainHandle(&mainTaskHandle);
}
```

coolwatcher trace:
[trace.zip](https://github.com/Ai-Thinker-Open/GPRS_C_SDK/files/5933767/trace.zip)

What is possibly wrong?

Thanks

glararan, 2021-02-06:
So i made little bit development.

Im afraid its circuit powering issue.

Used demo/call with following changes (to repeat call)
```
#include "stdbool.h"
#include "stdint.h"
#include "stdio.h"
#include "string.h"

#include "api_os.h"
#include "api_debug.h"
#include "api_event.h"
#include "api_call.h"
#include "api_audio.h"

#include "demo_call.h"

#define MAIN_TASK_STACK_SIZE    (2048 * 2)
#define MAIN_TASK_PRIORITY      0
#define MAIN_TASK_NAME          "Main Test Task"

#define SECOND_TASK_STACK_SIZE    (2048 * 2)
#define SECOND_TASK_PRIORITY      1
#define SECOND_TASK_NAME          "Second Test Task"

static HANDLE mainTaskHandle = NULL;
static HANDLE secondTaskHandle = NULL;

uint8_t flag = 0;
bool isDialSuccess = false;

void EventDispatch(API_Event_t* pEvent)
{
    switch(pEvent->id)
    {
        case API_EVENT_ID_NO_SIMCARD:
            Trace(10,"!!NO SIM CARD%d!!!!",pEvent->param1);
            break;

        case API_EVENT_ID_SYSTEM_READY:
            Trace(1,"system initialize complete");
            break;

        case API_EVENT_ID_NETWORK_REGISTERED_HOME:
        case API_EVENT_ID_NETWORK_REGISTERED_ROAMING:
            Trace(1,"network register success");
            flag = 1;
            break;
        case API_EVENT_ID_CALL_DIAL://param1: isSuccess, param2:error code(CALL_Error_t)
            Trace(1,"Is dial success:%d, error code:%d",pEvent->param1,pEvent->param2);
            if(pEvent->param1)
                isDialSuccess = true;
            break;
        case API_EVENT_ID_CALL_HANGUP:  //param1: is remote release call, param2:error code(CALL_Error_t)
            Trace(1,"Hang up,is remote hang up:%d, error code:%d",pEvent->param1,pEvent->param2);
            break;
        case API_EVENT_ID_CALL_INCOMING:   //param1: number type, pParam1:number
            Trace(1,"Receive a call, number:%s, number type:%d",pEvent->pParam1,pEvent->param1);
            OS_Sleep(5000);
            if(!CALL_Answer())
                Trace(1,"answer fail");
            break;
        case API_EVENT_ID_CALL_ANSWER  :  
            Trace(1,"answer success");
            break;
        case API_EVENT_ID_CALL_DTMF    :  //param1: key
            Trace(1,"received DTMF tone:%c",pEvent->param1);
            break;
[... 1327 lines cut ...]
[12:05:26.114]   L1S 02 : Tx Dummy on Tn 4 TxP 5\n
[12:05:26.142]   L1S 02 : Tx Dummy on Tn 5 TxP 5\n
[12:05:26.142]          : Fn 0195056 T1 0147 T2 04 T3 32 Time 00 02 08 916
[12:05:26.142]   L1S 02 : Tx Dummy on Tn 4 TxP 5\n
[12:05:26.224]   L1S 02 : Tx Dummy on Tn 5 TxP 5\n
[12:05:26.225]          : Fn 0195075 T1 0147 T2 23 T3 00 Time 00 02 08 999
[12:05:26.225]   PAL 12 : SynchChange[2500,168]
[12:05:26.225]   MM 09  : Dump : at 0x8209953D (5/5) : 32 f0 20 07 25
[12:05:26.225]   MM 09  : Dump : at 0x8209954E (6/6) : 32 f0 20 07 25 02
[12:05:26.225]   MM 09  : Dump : at 0x820994B7 (6/6) : 32 f0 20 07 25 02
[12:05:26.225]   MM 09  : Dump : at 0x820994C9 (6/6) : 32 f0 20 07 25 02
[12:05:26.326]   MM 09  : Att 1, Service 0x82, NMO 2
[12:05:28.229]          : Fn 0195510 T1 0147 T2 16 T3 27 Time 00 02 11 009
[12:05:28.341]   PAL 12 : SynchChange[0,126]
[12:07:40.248]          : Fn 0224114 T1 0169 T2 20 T3 20 Time 00 04 23 028
[12:07:40.358]   RRD 05 : rrd_FSM 0 WARNG: Evt 0x38081 unknown in rrd_VOID\n\n
```

My circuit scheme:
[image]

PS: red part and battery are not connected yet! Most parts are assembled by JLCPCB (except Ai thinker and sim)

Power inductor for powering ai thinker - https://lcsc.com/product-detail/Power-Inductors_SXN-Shun-Xiang-Nuo-Elec-SMMS0650-4R7M_C149582.html 
Its little bit noisy but it should do work.. or might not in my case?

jcherrabi, 2021-02-06:
did your call demo work on the A9G Putting module? would you like me to try it...?

glararan, 2021-02-07:
@jcherrabi dont have pudding module at the moment.. if you can try would be nice.

glararan, 2021-02-07:
So finally i got working demo/call after downloading zkemble v2.129.

I just tested out my original code on older circuit which had everything ok. Well results similiar = freeze.

zkemble, 2021-02-10:
This might give some clues into the power issues and calling problems https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421

> I replaced antenna https://lcsc.com/product-detail/Antennas_BAT-WIRELESS-BW433JWX50-10ZJ_C496575.html

That's a 433MHz antenna, not really meant for GSM frequencies, but it sounds like it works better than whatever you were using before :P

glararan, 2021-02-14:
So I can confirm its SDK bug. This friday I went deploy device in building made of metal (90%). My short antenna wasnt able to catch enough signal which resulted into crash. However as I moved it few centimeters away I was able to catch signal - no crash. In bad place we used larger antenna. 

However this issue should be opened. Unfortunately this SDK has no futher development :/