Ai-Thinker-Open/GPRS_C_SDK #480: File management with internal flash and tf card

breadcrumb3000, 2021-02-10:
## 1. SDK version（SDK 版本）

{

GPRS_C_SDK_v2129

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/demo/fs/

}

---
How to manage (copy / delete / edit) the files on the flash internal?  I would like to save some .txt and .mp3 files to the flash internal - how to copy files from a computer to the flash internal?

Is there also a way to copy files from computer to the TF card already inserted in the A9G, without removing the TF card from the A9G?

zkemble, 2021-02-10:
Internal flash is accessed at the root directory "/internal_data.txt", SD card storage is accessed in the /t/ directory "/t/sdcard_data.txt".
To transfer files between internal flash and your computer you will have to manually code the A9G to copy files between the internal flash and the SD card and use your computer to manage the files on the SD card. Unless you make your own protocol to transfer file data over UART.

https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/function-api/file-system.html

breadcrumb3000, 2021-02-11:
Thank you @zkemble .  It looks like the procedure to copy file from SD card to internal is: