Ai-Thinker-Open/GPRS_C_SDK #481: Non-standard UART baudrates doesn't work and break debug uart.

remixoff, 2021-02-10:
## 1. SDK version（SDK 版本）
V2129
{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
When non-standart integer value used as UART Speed for open UART1 (actually UART2 at the pins) all uarts became fail. Also HOST_TX/RX.
Reproduce:

open demo/uart project
change baudrate from UART_BAUD_RATE_115200 -> 25000
Build and flash
Expected behavior:
Uart1 works at 25000 baud, Host communication like tracker works
Actual:
Uart1 and 2 is free from data, Host communication recives a trash once then stop until reset the board. Reflashing board is possible only with way described for 'while(1) loops in the code' (i.e. start flashing in 1st second after reset.
}

---

zkemble, 2021-02-10:
UART_BAUD_RATE_25000 does not exist in the UART_Baud_Rate_t enum https://github.com/Ai-Thinker-Open/GPRS_C_SDK/blob/master/include/api_inc/api_inc_uart.h#L36

Building your project would have resulted with an error.

remixoff, 2021-02-11:
I know it. In embedded systems sometimes it needed to use non-usual baudrates. this is why enum you provided is actually set of consts. UART_BAUD_RATE_25000 is really absent, but try use integer constant 25000u instead. In ESP IDF I used this way successfully and reach real baudrate 25K. In the A9G SDK this fails. I guess this is bug in calculating dividers' values.

remixoff, 2021-02-11:
Upd: now I see that the UART_Baud_Rate_t really is a enum. Without equal signs (like UART_BAUD_RATE_115200 = 115200), but why? 
Using this baudrates set is just agreement, not a requirement. Even usb-ttl converters sometimes allows to set free speed, why something soldered to a PCB surprisingly limits the baudrate with several values?