Ai-Thinker-Open/GPRS_C_SDK #484: error: cust_rules.mk:577: recipe for target 'bin' failed  and cust_rules.mk:396: recipe for target 'lod' failed

naurych5758, 2021-02-20:
I've downloaded the GPRS_C_SDK and extracted it to c:\GPRS_C_SDK. Then I tried the build.bat using powershell:

PS C:\GPRS_C_SDK> .\build.bat demo first

but it produced error. Below are the outputs:

Windows PowerShell
Copyright (C) 2009 Microsoft Corporation. All rights reserved.

PS C:\Users\John> cd\
PS C:\> cd .\GPRS_C_SDK
PS C:\GPRS_C_SDK> .\build.bat demo first
CSDTK PATH: C:\CSDTK42
build folder exist
number of processors: 1
-- System Version : WINDOWS
---------------------------------------------------

[MAKE]  init

[AR]   libinit_debug.a

[MAKE]  libs

[MAKE]  libs/gps

[MAKE]  libs/gps/minmea

[AR]   libminmea_debug.a

[AR]   libgps_debug.a
        |
         -- libminmea_debug.a

[MAKE]  libs/utils

[AR]   libutils_debug.a

[MAKE]  libs/cjson

[AR]   libcjson_debug.a

[AR]   liblibs_debug.a
        |
         -- libgps_debug.a
        |
         -- libutils_debug.a
        |
         -- libcjson_debug.a

[MAKE]  demo/first

[AR]   libfirst_debug.a

-----------------------------------------------
[LDGEN]             >>> cust.ld
[LD] 1:C:/GPRS_C_SDK/build/first//first.elf 2:--oformat=elf32-littlemips 5:-just-symbols C:/GPRS_C_SDK/platform/chip/rom
/8955/lib/mem_bridge_rom_CHIP.elf 6: C:/GPRS_C_SDK/build/first/cust.ld 7:--gc-sections  -LC:/GPRS_C_SDK/build/first/init
/lib   -LC:/GPRS_C_SDK/build/first/libs/lib   -LC:/GPRS_C_SDK/build/first/demo/first/lib      --start-group   -linit_deb
ug  -llibs_debug  -lfirst_debug --end-group -LC:/GPRS_C_SDK/platform/lib/gcc/mips-elf/4.4.2 -lgcc  --warn-common  --erro
r-unresolved-symbols  8:C:/GPRS_C_SDK/build/first//first.map  end1             first.elf
[ElfCombine]        Elf binary & map file

[GEN]               strip platform symbol
                    |
                     --C:/GPRS_C_SDK/platform/compilation/platform_symbols_to_strip
mips-elf-ld.exe: hex/first/first.elf: access beyond end of merged section (-99877216)

[SREC]              first.srec for flash/romulator
53
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:577: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:396: recipe for target 'lod' failed
make: *** [lod] Error 2
-------------------------------------------------
ROM    total:1048576 Bytes     used:432 Bytes
RAM    total:1048576 Bytes     used:16 Bytes
=================================================
Start Time : 21:33:23.10
End   Time : 21:33:34.61
Build Time : 11.51s
=================================================

miguemorales, 2021-04-20:
You need visual redist 2010: https://www.microsoft.com/es-es/download/details.aspx?id=5555

oscarrock, 2022-03-01:
I have installed visual redist 2010 still having same error. Tryed on linux, tryed with docker... same issue. 

<pre>
PS C:\GPRS_C_SDK> .\build.bat app
CSDTK PATH: C:\CSDTK42
build folder exist
number of processors: 8
-- System Version : WINDOWS
---------------------------------------------------

[MAKE]  init

[AR]   libinit_debug.a

[MAKE]  libs

[MAKE]  libs/gps

[MAKE]  libs/gps/minmea

[AR]   libminmea_debug.a

[AR]   libgps_debug.a
        |
         -- libminmea_debug.a

[MAKE]  libs/utils

[AR]   libutils_debug.a

[MAKE]  libs/cjson

[AR]   libcjson_debug.a

[AR]   liblibs_debug.a
        |
         -- libgps_debug.a
        |
         -- libutils_debug.a
        |
         -- libcjson_debug.a

[MAKE]  app

[AR]   libapp_debug.a

-----------------------------------------------
[LDGEN]             >>> cust.ld
[LD] 1:C:/GPRS_C_SDK/build/app//app.elf 2:--oformat=elf32-littlemips 5:-just-symbols C:/GPRS_C_SDK/platform/chip/rom/8955/lib/mem_bridge_rom_CHIP.elf 6: C:/GPRS_C_SDK/build/app/cust.ld 7:--gc-sections  -LC:/GPRS_C_SDK/build/app/init/lib   -LC:/GPRS_C_SDK/build/app/libs/lib   -LC:/GPRS_C_SDK/build/app/app/lib      --start-group   -linit_debug  -llibs_debug  -lapp_debug --end-group -LC:/GPRS_C_SDK/platform/lib/gcc/mips-elf/4.4.2 -lgcc  --warn-common  --error-unresolved-symbols  8:C:/GPRS_C_SDK/build/app//app.map  end1             app.elf
[ElfCombine]        Elf binary & map file

[GEN]               strip platform symbol
                    |
                     --C:/GPRS_C_SDK/platform/compilation/platform_symbols_to_strip

[SREC]              app.srec for flash/romulator
53
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:577: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:396: recipe for target 'lod' failed
make: *** [lod] Error 2
-------------------------------------------------
ROM    total:1048576 Bytes     used:704 Bytes
RAM    total:1048576 Bytes     used:48 Bytes
=================================================
Start Time : 16:45:18.75
End   Time : 16:45:24.75
Build Time : 6.0</pre>

mishagreen55, 2023-07-31:
I installed the Visual C redistributables, but still have the problem. 

Line 577 in cust_rules.mk has been commented out: the error occurs on line 597.

`	597: srecmap -c ${MAP_FILE} -m ${FLSH_MODEL} -b ${TARGET_FILE} ${HEX} ${LODBASE} ${STDOUT_NULL}
`
Turning on debug in make, this is the command that fails:
```
CreateProcess(C:\proj\A9GSM\CSDTK4~1\CSDTK42\make64\sh.exe,C:/proj/A9GSM/CSDTK4~1/CSDTK42/make64/sh.exe -c "srecmap -c C:/proj/A9GSM/GPRS_C_SDK-master/platform/compilation/8955_map_cfg -m flsh_spi32m -b C:/proj/A9GSM/GPRS_C_SDK-master/build/gpio/targetgen C:/proj/A9GSM/GPRS_C_SDK-master/build/gpio//gpio.srec C:/proj/A9GSM/GPRS_C_SDK-master/hex/gpio/gpio_ 1>dev/null",...)
Live child 00000000026D5530 (bin) PID 40753936
53
Reaping losing child 00000000026D5530 PID 40753936
C:/proj/A9GSM/GPRS_C_SDK-master/platform/compilation/cust_rules.mk:578: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
```
the STDOUT from srecmap is routed to /dev/null: if you route it to a file instead, the file is created but is empty.

rganesh1982, 2025-01-21:
> You need visual redist 2010: https://www.microsoft.com/es-es/download/details.aspx?id=5555

This worked for me, Thank you!