Ai-Thinker-Open/GPRS_C_SDK #486: Wrong Coordinates

msevimli, 2021-04-05:
## 1. SDK version（SDK 版本）

{

csdtk42

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Hello there, ,I've serious problem here, everything seems working, i can get phone call, i can send and receive sms but when i want to use i am getting wrong coordinate, if plug out sim card gnd for try, it says : "wait for gprs register complete" at my coolwatcher Trace section, when i plug in it again everything seems working but i am getting wrong coordinate as : 

GPS fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:0, gps sates total:1, is fixed:no fix, coordinate:WGS84, Latitude:90.000000, Longitude:0.000000, unit:degree,altitude:0.000000

I couldn't figure out this problem for weeks, any solution or any way to close to problem i'd be really appricated
Thanks 

}

---

zkemble, 2021-04-05:
>GPS fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:**0**, gps sates total:**1**, is fixed:**no fix**

It's telling you that it has no fix because it can only see 1 satellite and tracking 0. It needs to track 3 or 4 satellites to get a fix, which can take a few minutes depending on signal quality. Make sure the GPS antenna has clear view of the sky.

msevimli, 2021-04-05:
Hi @zkemble , 
Thanks for your quick response, even if wait hours there is no success, i have pudding dev board and if i connect same antenna to pudding with same wire and same place, it takes a few minutes to give correct coordinates, on my curcuit ( on breadboard ) i don't have voltage regulator ( SY8089 ) and 2.2UH inductor yet, i am using adj Power Supply for
set Voltage : 4.20 volt 
current : 00.05 A ( sometimes up to 00.12 ) 
i am really  stuck on this and i can't go to next step

zkemble, 2021-04-05:
GPS antennas are usually active antennas with a built-in amplifier. They need power which is usually supplied through an inductor - L2 82nH on the pudding dev board schematic. Have you included that in your setup? Best post a schematic or something.

msevimli, 2021-04-05:
Hi again, 
@zkemble  Many thanks for your valuable time and attentions 
I've included L2 82nh inductor and tried two different 
one is :
https://lcsc.com/product-detail/HF-Inductors_EMTEK-HQC0805-82NJ-T_C251939.html

second :
https://lcsc.com/product-detail/Inductors-SMD_Chilisin-Elec-CS1008-82NG-S_C285940.html

and also i tried with pudding original antenna and this :
https://lcsc.com/product-detail/Antennas_Rainsun-microwave-Tech-GPS1003_C239243.html

result is the same, i will attach my schematic also

msevimli, 2021-04-05:
Hi @zkemble 
And this is my schematic 
[image]

zkemble, 2021-04-05:
The GPS antenna you linked to is a passive antenna, you might need a GPS LNA between the antenna and A9G GPS input (remove the inductor if you're not using an external antenna).
You don't have any decoupling capacitors on your power supply, very important, weird things happen with dodgy power - https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/421

msevimli, 2021-04-05:
Thanks for quick reply and your attentions 
I will try to edit as your mentions and let you know  to result 
I hope it will solve

msevimli, 2021-04-06:
Hi again @zkemble,

After your valuable information and advise  i spend over two nights and working with this, finally i could succeed to get coordinates,
I added 3 capacitor to VBAT, i am attaching latest schematic,

[image]

When i done this first time with external active antenna ( pudding original antenna ) it was really amazing output as :

_GPS fix mode:3, BDS fix mode:3, fix quality:1, satellites tracked:4, gps sates total:15, is fixed:3D fix, coordinate:WGS84, Latitude:55.5XXXXX, Longitude:13.0XXXXX, unit:degree,altitude:0.000000_ 

and it was the most correct coordinates with only a few meters difference than real, 

Then after a problem was arised with same circuit  connections and same antenna's  place
 I couldn't get the same result, the output as 

_GPS fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:0, gps sates total:1, is fixed:no fix, coordinate:WGS84, Latitude:55.5XXXXX, Longitude:13.0XXXXX, unit:degree,altitude:0.000000_

Coordinates are close to first one but this time difference is around 120 meters  and when gps states total increase about 2-3 it seems to reset itself with: 

```
[00:57:18.272]   MMI 02 : GPS fix mode:1, BDS fix mode:1, fix quality:0, satellites tracked:0, gps sates total:2, is fixed:no fix, coordinate:WGS84, Latitude:55.5XXXXX, Longitude:13.0XXXXX, unit:degree,altitude:0.000000
[00:57:18.346]          : Fn 0796065 T1 0600 T2 23 T3 06 Time 00 20 25 074
[00:57:18.346]   MM 09  : Dump : at 0x8209953D (5/5) : 42 f0 42 ff fe
[00:57:18.346]   MM 09  : Dump : at 0x8209954E (6/6) : 42 f0 42 27 75 00
[00:57:18.347]   MM 09  : Dump : at 0x820994B7 (6/6) : 42 f0 42 27 75 ff
[00:57:18.347]   MM 09  : Dump : at 0x820994C9 (6/6) : 42 f0 42 27 75 00
[00:57:18.347]   MM 09  : Att 1, Service 0x01, NMO 256
[00:57:18.347]   MM 09  : return for PrioAccThr\n
[00:57:18.347]   CSW 02 : CFW_ShellProc [nEvId]0x138d,[nParam1]0x3 [nSimID]0,[nType]02\n
[00:57:18.358]   CSW 02 : Limited Server\n
[00:57:18.358]          : Fn 0796068 T1 0600 T2 00 T3 09 Time 00 20 25 086
[00:57:18.458]   PAL 12 : SynchChange[0,130]
[00:57:19.051]          : Fn 0796218 T1 0600 T2 20 T3 06 Time 00 20 25 779
[00:57:19.151]   RRI 05 : rri_P 0 WARNG: Evt 0x38015 unknown in rri_PLTD\n\n
[00:57:23.188]          : Fn 0797114 T1 0601 T2 06 T3 35 Time 00 20 29 916
```
i tried to add one more 470 uf Capacitor but the result is same.. 
May check it please is there anything that i missed ?
Thanks for your all patience and efforts