Ai-Thinker-Open/GPRS_C_SDK #488: How can I set the SIM pin code for GPRS access

Koxx3, 2021-04-28:
I looked at the SDK API, and I din't find any answer.
Any help please ?

rolonja, 2021-06-02:
i have the same issue

faithware, 2021-08-14:
You can't, there is no implementation for that unless you are using AT commands

MadDogMayCry0, 2022-01-24:
@faithware 
Hi!
Can we execute AT commands through SDK? Or do we only can use external MCU for that?

faithware, 2022-01-24:
Not that I know of. @MadDogMayCry0

MadDogMayCry0, 2022-01-24:
@faithware 
You mentioned in one of the threads that a95 is a problematic chip. Could you highlight the most obvious ones that do not allow using it in a project? And maybe you could recommend something else?

faithware, 2022-01-24:
The software development for the rda8955 has been stopped, contacting Ai thinker or unisoc about all the problems gets you nowhere. Also u never know when the A9 goes out of stock forever. 
If u can tolerate these things then you may suffer a bit and get things to work.