Ai-Thinker-Open/GPRS_C_SDK #494: GPS data transmission stops after 1 day

cityba, 2021-07-13:
I run the gps-tracker code, rewriting it a bit, and 24 hours after turning it on, the data transfer stops, the sim card goes, it rings. What could be wrong? why does it stop after exactly 1 day? is there any hidden function that causes this? release code also stops at 24 hours.Can anyone help?

breadcrumb3000, 2021-07-31:
have you tried building your app using "./build.bat demo $PROJ release" to build in "release" mode instead of "debug" mode?
Source is here : https://www.iot-experiments.com/ai-thinker-a9-module-with-custom-firmware/

faithware, 2021-08-14:
> @xuhongv @punee995 @khaledbenh @vencol @a710450641 @zerosigu @ZhongShangWen @fengfeng0328 I run the demo_gps-tracker code, rewriting it a bit, and 24 hours after turning it on, the data transfer stops, the sim card goes, it rings. What could be wrong? why does it stop after exactly 1 day? is there any hidden function that causes this? release code also stops at 24 hours.Can anyone help?

the issue is that if gprs connection goes down, you can't reconnect unless you reboot the A9.
There is a workaround only if your sim provider allow gprs connection with a known APN

breadcrumb3000, 2021-12-12:
Watchdog?

khanhcq888, 2022-02-21:
build release file and upload. it is ok. 
I have a same problem and I tried to build release version, so GPS run in several day continuos

cityba, 2022-02-21:
The solution is to add the reconnection to the gprs break event code.
API_EVENT_ID_NETWORK_REGISTER_DENIED
API_EVENT_ID_NETWORK_DETACHED
API_EVENT_ID_NETWORK_ATTACH_FAILED
API_EVENT_ID_NETWORK_DEACTIVED
API_EVENT_ID_NETWORK_ACTIVATE_FAILED

`...case API_EVENT_ID_NETWORK_DETACHED:
        Trace(1,"network detached");            
        AttachActivate();
        break;.....`