Ai-Thinker-Open/GPRS_C_SDK #495: sdk lod bin fail a9g

joehaughey, 2021-07-27:
## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

}

---

joehaughey, 2021-07-27:
csdtk42 recursive

./buid.bat demo gps

C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:577: recipe for target 'bin' failed
make[1]: *** [bin] Error 181
C:/GPRS_C_SDK/platform/compilation/cust_rules.mk:396: recipe for target 'lod' failed
make: *** [lod] Error 2

joehaughey, 2021-07-27:
is there a cust_rules.mk entry that is required

breadcrumb3000, 2021-12-12:
are you using Powershell?

kadilab, 2022-08-01:
Bonjour j'ai aussi le même problème avez vous la solution ?

reydevelop, 2022-12-14:
Any solution yet?

kicsigabszi, 2022-12-22:
this solution worked for me
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/180#issuecomment-409844287