Ai-Thinker-Open/GPRS_C_SDK #498: GPS satellites

JazminFerre, 2021-09-25:
Hi,
I'm testing the GPS mode, with the GPS demo code, and it works. However, it must be connected to 9 satellites to get the correct location. I think it should work with 3 satellites connected. Can somebody help me?
The location where I am is quite clear.
Thanks!!!!
Greeting,
Jaz

zkemble, 2021-09-25:
4 satellites are the minimum required for a fix. The position of the satellites is also important; if it has a lock on 4 and they're all bunched together in the same area of the sky then the receiver will have a hard time working out its position. The more satellites locked the quicker its position can be determined. Your module likely had to lock to 9 satellites before it found some that were spread out enough and had a good enough signal before it could acquire a fix.

JazminFerre, 2021-09-28:
OK thank you!! That was a pretty clear explanation.

closed 2021-09-28 by JazminFerre