Ai-Thinker-Open/GPRS_C_SDK #506: SD card access

willemneethling, 2022-01-16:
## 1. SDK version（SDK 版本）

{
N/A
}

Is there any way to access the SD card without resorting to custom firmware, e.g. via AT commands? There is a unsoldered header on the A9G "pudding" dev board which seems to relate to the SPI interface to the SD card, has anyone ever been able to use that?

Mukund2900, 2024-06-12:
@willemneethling did you find anything for this ?
I have connected esp32 to a9g but esp32 is unable to read/write/connect to sd card on a9g. 
Is there a way to do it

willemneethling, 2024-06-13:
@Mukund2900 No luck. I did not find any documented way of using the SD slot without resorting to the custom chinese SDK. I also soldered a header onto the exposed SD pins and tried to drive this from the ESP32 without success.