Ai-Thinker-Open/GPRS_C_SDK #510: NMEA data is mixed into UART1 data during transmission

khanhcq888, 2022-02-21:
## 1. SDK version（SDK 版本）

{

V2129

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Hello everyone!

I have a problem with A9G when I receive the data via UART1. The NMEA data is mixed into UART1 data during transmission. I tried everything but the NMEA data always mixed into UART1 data. 
Can anyone help me to resovle problem with A9G? Thanks all!

This is the example about the NMEA is mixed into UART1 data.

data:image/jpeg;base64,%2F9j%2F4AAQSkZJRgABAQEAAAAAAAD%2F2wBDAAwICQsJCAwLCgsODQwOEh4UEhEREiUaHBYeLCYuLSsmKikwNkU7MDNBNCkqPFI9QUdKTU5NLzpVW1RLWkVMTUr%2F2wBDAQ0ODhIQEiMUFCNKMioySkpKSkpKSkpKSkpKSkpKSkpKSkpKSkEQACAQIEBAMEBwUEBAABAncAAQIDEQQFITEGEkFRB2FxEyIygQgUQpGhscEJIzNS8BVictEKFiQ04SXxFxgZGiYnKCkqNTY3ODk6Q0RFRkdISUpTVFVWV1hZWdec0WFcpSy6ShMct%2FFyCrDJGK8u000.0000,E,0,0,,137.0,M,13.0,M,,*57
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$BDGSA,A,1,,,gRgilqSgpaYDulIakQtHSgBaOlIYZopiCilYYUUwDj0ooEFFAxpNLQAlNoAWk7UgEpeKYBikoASigBKOKQCUUwGNTaAE704dRQBYeoDQABa7bwdL5mhuh6wzbf0qZCOp0ziGb8f%2FQaxvl2jntWJQg61xPids%2BIr%2FwD66VcfiAzhU8VWxCyciq5GKEAoNO8yqsBLFNNkVqXbfVHkV1uYd7fwmMbazn1N2YbEKNjB3AGlYBwurjO8Fc%2FwC4KfJf3kiFXeMj%2FrnU6FWId0v9805bm5QY8wEe6U7jJ01C8HSSP%2Fv3TjqF4%2FDmBgf%2BmIqdA
$GPGSV,1,1,00*79
$BDGSV,1,1,00*68
$GNRMC,000129.259,V,9000.0000,N,00000.0000,b2MD29nDFJ96NNtdHQzaLOOKaRQIbSGqCw2kpCGEcVFIMrimyrHJ3i%2Fv39M1BWa

}

---

breadcrumb3000, 2022-03-03:
Are you connected to the correct pins?

khanhcq888, 2022-03-03:
> Are you connected to the correct pins?

Yes, of course. I connect to UART1. When I turnoff GPS, the data is ok. But when turn on, the NMEA jumps into data