Ai-Thinker-Open/GPRS_C_SDK #515: A9 stops working after several resets

faithware, 2022-06-20:
I have been using the A9 chip for a good 4 years.
I have circumvented several issues with the SDK but I can't find the solution for stopping.
When power is down and the battery is running low, the A9 chip will reset all the time until it stops and can't work again  until the device is reflashed.
`[12:03:17.357]          : Detected event: 0xffffffff
[12:03:17.449]          : Detected event: 0xf07a0000
[12:03:17.489]          : Detected event: 0xf07a0002
[12:03:17.489]          : Detected event: 0x89550002
[12:03:17.499]          : Detected event: 0xfe900800
[12:03:17.499]          : Detected event: 0xfe901b00
[12:03:17.519]          : Detected event: 0x00000004
[12:03:17.520]          : Detected event: 0x66660021
[12:03:17.520]          : Detected event: 0x0000000f
[12:03:17.520]          : Detected event: 0x00000000
[12:03:17.527]          : Detected event: 0x001660c8`

garudaonekh, 2022-06-22:
@faithware do you figure out why? I faced this same issue as well which I just reflashed it.

peterjschulze, 2022-06-22:
Hi All We have about 20 devices in the field for over 2 years now.We connect the power supply(mains) to the usb connector and a battery to the battery connections on the board, we have no issues. If the battery voltage drops below 3.2v the device switches off and restarts without issue when the usb power returns. Thank you, hope this helps. Peter Sent from Mail for Windows From: Gnichi MohamedSent: Monday, 20 June 2022 10:47To: Ai-Thinker-Open/GPRS_C_SDKCc: SubscribedSubject: [Ai-Thinker-Open/GPRS_C_SDK] A9 stops working after several resets (Issue #515) I have been using the A9 chip for a good 4 years.I have circumvented several issues with the SDK but I can't find the solution for stopping.When power is down and the battery is running low, the A9 chip will reset all the time until it stops and can't work again until the device is reflashed.[12:03:17.357] : Detected event: 0xffffffff [12:03:17.449] : Detected event: 0xf07a0000 [12:03:17.489] : Detected event: 0xf07a0002 [12:03:17.489] : Detected event: 0x89550002 [12:03:17.499] : Detected event: 0xfe900800 [12:03:17.499] : Detected event: 0xfe901b00 [12:03:17.519] : Detected event: 0x00000004 [12:03:17.520] : Detected event: 0x66660021 [12:03:17.520] : Detected event: 0x0000000f [12:03:17.520] : Detected event: 0x00000000 [12:03:17.527] : Detected event: 0x001660c8

Vetal7979, 2024-01-25:
was there any solution? I have a device powered from a 12V car socket that periodically works exactly the same way. Without flashing, further work is impossible