Ai-Thinker-Open/GPRS_C_SDK #516: Stability and service life

garudaonekh, 2022-06-22:
hi,
Can you share your experience on this module? I love its feature(2g gprs+gps+soc) but I doubt its stability for deploying to iOT projects. I need it for Farm iot projects. Our minimum requirement is 2 years+ service life. 
I am testing it on a project for 3+ months now and no issue. 

Thanks

niwantha33, 2022-06-22:
Hi, I  used this module in  2020  to build a small IoT project sending meter readings to IoT Cloud, so I observed that the module needs to reset from time to time using an external device(another MCU) and also sometimes does not connect to the server. I think it is better to find some solution in the case mentioned errors occurred to your system.

garudaonekh, 2022-06-22:
> Hi, I used this module in 2020 to build a small IoT project sending meter readings to IoT Cloud, so I observed that the module needs to reset from time to time using an external device(another MCU) [...]

Thanks for your response. Did you investigate further the main cause? Is it due to hardware or software issue? Internet connection stability is kind of difficult thing to maintain 100% uptime.

garudaonekh, 2022-06-23:
> Having an external mcu is a must because when the gprs disconnects there is no way to reconnect unless you reset the module. Also u need to monitor the power supply to the A9, it doesn't like resetting too much. Overall, The module itself has been working for me since 2019 and I have 150 devices in the field that work for straight two years

Thanks. Among the 150 devices, how many failed so far? 

I used micropython which I have to reset the device every two days to avoid mqtt connection issue which hanged the device and watchdog comes to the rescue. GPRS reconnect work after several retries in micropython.

faithware, 2022-06-23:
@garudaonekh maybe 3 or 4 died because of A9 failure. The Mqtt connection on my part can last up to two weeks max (I am not using micropython btw) and gprs reconnect doesn't work because the SIM cards that I use don't require a valid apn to work.

Benedito821, 2023-11-28:
@faithware did you have to pay for a mobile data packet for the GPS to work properly? Is it suitable for urban applications where there are high buildings?

garudaonekh, 2023-11-29:
> @faithware did you have to pay for a mobile data packet for the GPS to work properly? Is it suitable for urban applications where there are high buildings?

GPS is independent chip and it is free so it is not related with GPRS data package that you have to pay to Telecom. 

However, its GPS chip is very weak and accuracy is bad. I think it is up to 50 to 100 meter errors and may not work inside a tall building

Benedito821, 2023-11-29:
Thank you @garudaonekh . Did you have troubles getting the coordinates in real-time? It seems that  it sometimes loses the connection with the satellites. Did a 2G module really match your needs? I think I'll have to go for a 4G module.

faithware, 2023-11-29:
It is very slow to start, it needs sometimes 5 minutes to track 4
satellites. It is not good enough honestly

garudaonekh, 2023-11-29:
What @faithware said is right. Its hot fix is very slow. Good enough or not depending on your use case. If you just need to know a rough location, then it fits the job with the price. 2G is dead in some countries. Please note that in some developing countries, 2G and SMS is still the best choice because of its long coverage in rural area. If you just need to send data occasionally with small payload, then it's ok.