Ai-Thinker-Open/GPRS_C_SDK #532: New A9G Board Design - Power Supply

ricardodanielalbano, 2023-05-03:
Hello, I'm designing a new board based on A9G module, I'm so confused about power supply for the module and li-ion charging for the battery. I see in the dev board A9G that there isn't carge circuit for the li-ion battery, it's connected directly to a dc-dc step down to 4.2v, also I'm confused if I need to supply 5v to VUSB pin in the A9G module if I'm using VBAT and charging it with the apropiate electronics.
Thank you for your help.
Regards,
RDA.-

Trion, 2023-05-09:
You can use tp4056 charger module ouput is 4.2v , A9G module max input voltage is 4.2 v , don't directly connected with 5v to A9G module , VBAT pin max input voltage is 4.2v , VBAT pin is INPUT pin for power supply, not a output pin,

Trion, 2023-05-09:
[image]
recommended  power supply design for A9G module,

ricardodanielalbano, 2023-06-13:
Hello, I need a datasheet for A9G module, I can't find it.
Need information about VIO,VUSB, USB_N and USB_P pins.
Regards,
RDA.-

Deenchi1, 2023-07-26:
> [image] recommended power supply design for A9G module,

please help i am unable to power my A9 with 18650 i just attached it to Vbat power for 2s and the A9 gives me back random responses for example: ejdkjdod3okdw

ricardodanielalbano, 2023-07-31:
Feeding 4.2v directly to your battery from the switching regulator is not good idea, also in this way the regulator can't do the job was designed for. You need to leave the battery uncoupled from your VBAT net while the switching regulator is working, in the other hand, when no dc input is present in the switching regulator you have to couple the battery again to vbat net, I have implemented this switching circuit in my design, you can check it at #538