Ai-Thinker-Open/GPRS_C_SDK #537: GPRS Module restart

ricardodanielalbano, 2023-06-16:
Please, any clue about restarting gprs module without restarting the whole device!.
In some cases GPRS get deatached from network and is impossible to reconnect it without restarting the module.
I know is a common issue but no solution is available as I can see.

ZakKemble, 2023-06-16:
I think `PM_Restart();` will restart the module without a power cycle.

ricardodanielalbano, 2023-06-16:
Thank you, but not. PM_Restart just Reboot the system.

jackfitton112, 2023-07-27:
think you need Network_StartDetach() and attach, it doesnt turn the modem off but it will disconnect from the network and lower power cosumption

MaBarakat, 2024-06-01:
@ricardodanielalbano are find a way please?

ricardodanielalbano, 2024-06-02:
No @MaBarakat, just I restart the device when no network is available for a while.

MaBarakat, 2024-06-02:
thank you very much

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
خالٍ
من الفيروسات.www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>