Ai-Thinker-Open/GPRS_C_SDK #538: Any recomendation for this board ?

ricardodanielalbano, 2023-06-17:
Hello, I'm working on a gps tracker based on A9G, we're planning to release this hardware and make some units soon.
Any suggestion about this design ? We're open to share ideas and experiences with de community. Best regards!.

[image]

jackfitton112, 2023-07-27:
As someone who made a gps tracker with this device i would advise against it. your schematic looks okay (you may want EMI potection on simcard to board) 

The documentation for the SDK is terrible, the board only supports GPRS / 2G which is being phased out, the SDK / flashing tools are very buggy.

proceed at your own risk..

valinet, 2023-07-29:
> As someone who made a gps tracker with this device i would advise against it. your schematic looks okay (you may want EMI potection on simcard to board)
>
> The documentation for the SDK is terrible, [...]

Do you have a recommendation for a different product? Especially looking for something that has low power consumption, cell connectivity (2G or 4G), GPS, built-in accelerometer would be a plus...

ricardodanielalbano, 2023-07-31:
> As someone who made a gps tracker with this device i would advise against it. your schematic looks okay (you may want EMI potection on simcard to board)
>
> The documentation for the SDK is terrible, [...]

Poor documentation is a true, I'm working with this module from several months, I have designed the PCB and making it in production, lot's of issues with code but finally get an stable firmware, the flasher/debbuger is a bit buggy, yes, but it works.
The module seems robust, you need to keep attention to get vbat stable source with enough current to feed the device with current peaks demands.

ricardodanielalbano, 2023-07-31:
> Do you have a recommendation for a different product? Especially looking for something that has low power consumption, cell connectivity (2G or 4G), GPS, built-in accelerometer would be a plus...

Also I'm looking for a 4G version of this module. I you have any clue please post it here!.

ricardodanielalbano, 2023-07-31:
Here is the finished board, with some modifications from the original schematic posted in this thread. The 4.2v switching regulator has reemplaced with a LDO and the inductor has incorrect value, I't must be about 220uH+ for this design. Also this design has issues with the load switching between bat/dc input, exesive conmutation time when no DC input is present makes the module power down and restart.
[image]