Ai-Thinker-Open/GPRS_C_SDK #549: at+location = 2 in a9g

carlestecno, 2024-01-13:
at+location=2

## 1. SDK version（SDK 版本）

{

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{
Sometimes in the same place, after turning off and on the board a9g the GPS, they connect quickly, sometimes the GPS takes a long time to connect, and sometimes it does not connect at all. Is there a way to fix this in the code? Thanks.

}

---

carlestecno, 2024-01-13:
This is part of my code:
`#include <SoftwareSerial.h>

SoftwareSerial mySerial(16, 17);

#define DEBUG true

String check_gps;
String check_coordinates;
String check_att;
String check_att_dos;
String check_conn;
String check_pdp;

int count_gps = 0;
int count_coordinates = 0;
int value_boton = 0;
int flag = 0;

String coordinates, lat, lng;
String mac = "a9g";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  mySerial.begin(9600);
  pinMode(4, INPUT);
  sendData("AT+RST=2", 1000, DEBUG); // per posar el mòdul en consum baix
  //check_gps = sendData("AT+GPS=1", 1000, DEBUG);
  //sendData("AT+GPSRD=20", 1000, DEBUG);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (check_gps.indexOf("OK") != -1)
  {
    count_gps = 1;
    Serial.println("GPS connectat");
  }
  else
  {
    Serial.println("Intentant connectar-se");
    check_gps = sendData("AT+GPS=1", 1000, DEBUG);
  }
  if (count_gps == 1)
  {
    if (count_coordinates == 0)
    {
      coordinates = sendData("AT+LOCATION=2", 2000, DEBUG);
      Serial.print("at + location: ");
      Serial.println(coordinates);
      if(coordinates.indexOf("OK") != -1)
      {
        count_coordinates = 1;
      }
    }`

carlestecno, 2024-01-13:
This is part of the code:

`#include <SoftwareSerial.h>

SoftwareSerial mySerial(16, 17);

#define DEBUG true

String check_gps;
String check_coordinates;
String check_att;
String check_att_dos;
String check_conn;
String check_pdp;

int count_gps = 0;
int count_coordinates = 0;
int value_boton = 0;
int flag = 0;

String coordinates, lat, lng;
String mac = "a9g";

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  mySerial.begin(9600);
  pinMode(4, INPUT);
  //sendData("AT+RST=2", 1000, DEBUG); // per posar el mòdul en consum baix
  check_gps = sendData("AT+GPS=1", 1000, DEBUG);
  //sendData("AT+GPSRD=20", 1000, DEBUG);
}

void loop() {
  // put your main code here, to run repeatedly:
  if (check_gps.indexOf("OK") != -1)
  {
    count_gps = 1;
    Serial.println("GPS connectat");
  }
  else
  {
    Serial.println("Intentant connectar-se");
    check_gps = sendData("AT+GPS=1", 1000, DEBUG);
  }
  if (count_gps == 1)
  {
    if (count_coordinates == 0)
    {
      coordinates = sendData("AT+LOCATION=2", 2000, DEBUG);
      Serial.print("at + location: ");
      Serial.println(coordinates);
      if(coordinates.indexOf("OK") != -1)`

carlestecno, 2024-01-13:
I will try outside. I'm close to the window  but perhaps I need to be outside. Thanks

carlestecno, 2024-01-14:
I think the problem was to be inside and not outside

Trion, 2024-08-06:
Gps are not working in building or inside room, 
According to the wiki :
"GPS signals are unable to pass through dense materials such as concrete, heavy wood or steel cladding.  To get the best effects from GPS tech, you really need visibility of a large area of sky to ensure that the received signal is strong enough pinpoint your location with any degree of accuracy"