Ai-Thinker-Open/GPRS_C_SDK #550: a9g and esp32 power supply

carlestecno, 2024-01-15:
Hi @faithware

I'd like to know how can I power on an a9g and an esp32 ( for gps tracker ) with power supply because I'm trying wiht battery 18650 3.7V and I can't power on. I saw you have some installed.

Thanks in advance.

carlestecno, 2024-01-15:
Solved pushing power key button. But how I can power on without power key?.

Trion, 2024-08-06:
you can connect VSB pin to 5v , 
Here the list of powersupply features:
**Power Supply**

1. •	board can be light up with 5V(with DC-DC on dev-board) or 4.2V(VBAT) power, and ensure supply enough current(Peak current 2A)
2. •	If use lithium battery connected to VBAT, you must press power-key for a while to power up the module, or just pull down the power-key pin.
3. •	**If VUSB connected with 5V (VBAT must supply with power),** the module will automatically power up no matter power-key poll down or not.