Ai-Thinker-Open/GPRS_C_SDK #562: A9G Firmware download bricked HST CHST_Read32 CoolHostTimeoutError

adeliktas, 2024-10-13:
## 1. SDK version（SDK 版本）

{

CSDTK 4.2 ($Revision: 43828 $)
GPRS_C_SDK current/upstream version https://github.com/Ai-Thinker-Open/GPRS_C_SDK
firmwarw_tool_v2.1

}

---

## 2. In what kind of operation problems appear, and how to reproduce the problem ?（什么样的操作步骤问题会出现，是否是稳定复现，如何复现问题？）

{

Under Windows 10 I tried to flash app and demo_uart and it did progress flashing to the end and gave some Triangle Warning for checksum?? But can't tell whether it succeeded. Constant Low on GPIO_PIN25 denies app running and no UART/Trace denies that example to be running either...
Whenever i flashed i always selected flash-programmer "cooltools/chipgen/Modem2G/toolpool/plugins/fastpf/flash_programmers/host_8955_flsh_spi32m_ramrun.lod"
and fastpfization LOD file GPRS_C_SDK/app (and GPRS_C_SDK/uart) big filesize lod file first then again with small size lod file of it
I also tried flashing mircropython firmware_debug_full.lod and firmware_debug_flash.lod but coolwatcher returned an error about incompatible or something and there is no python console in serial.
I have double checked cabling, pinouts and used two separate A9G Modules and USB TTL (CH340G, CP2102).
Of course: TX->HST_RX, RX->HST_TX, GND->GND, 5V->5V
I still receive serial RX/TX 115200 with AT Commands works ...
Trace Tool is empty and doesn't receive anything
I have followed the documentations and looked up many reported issues and guides
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/installation.html
https://ai-thinker-open.github.io/GPRS_C_SDK_DOC/en/c-sdk/burn-debug.html
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/140
https://github.com/Ai-Thinker-Open/GPRS_C_SDK/issues/249
https://techiesms.com/steps-for-flashing-micropython-firmware-on-a9g/
I did successfully flash the latest firmware (version confirmed with AT+CGMR) from https://docs.ai-thinker.com/en/gprs/firmware ai-thinker_a9_a9g_at_v02.02.20190915r.rar with the tool https://docs.ai-thinker.com/gprs_download but i did not include gps firmware to the upgrade since the link was dead and i was just testing.
Also note when i press and hold the PWR Button for 1s IO28/LED2 is permanent blue until i press reset and i couldn't find what that means anywhere.
The Device/COM Port does not show up anymore in the Ai-Thinker Firmware Download Tool either.
Reflashing Firmware is not possible like this.
To be clear the issue is with A9G Device/Firmware. I have tried many setups and excluding errors.

Simple steps for me to reproduce:
1. connect A9G to USB TTL to PC.
2. open firmware tool and it will not show up no matter pressen RST/PWR Buttons
3. OR open cooltools coolwatcher and try to flash any LOD gives CoolHostTimeoutError 
[image]

}

---

navidbaghaee, 2024-11-24:
I have the same issue,
Do you have any progress on this problem?

adeliktas, 2024-11-24:
> I have the same issue, Do you have any progress on this problem?

i bought 4 in total and only the one got bricked. I've tried everything now (all kinds of usb ttl chips, suggestions etc.) messing around with it too much and accidently reversed the GND and 5V smoking it, but the problem i assume likely was with the custom C example firmware code not having a failsafe/kill-switch (or sleep at least according to docs?) that you definitely want to add even with other firmware like micropython since you will have issues to break out of the running firmware environment (takes a few restarts/retries to exit any program in micropython without failsafes). No issues with the other 3 a9gs running micropython and for the fail-safe i added code connecting PinX to GND will quit the program/return to the micropython interpreter.